AtsushiSakai/PythonRobotics #546: EKFSLAM ipynb issue

Ganeshgng, 2021-09-26:
https://github.com/AtsushiSakai/PythonRobotics/blob/master/SLAM/EKFSLAM/ekf_slam.ipynb

Got a warning
usr/local/lib/python3.7/dist-packages/ipykernel_launcher.py:27: VisibleDeprecationWarning: Creating an ndarray from ragged nested sequences (which is a list-or-tuple of lists-or-tuples-or ndarrays with different lengths or shapes) is deprecated. If you meant to do this, you must specify 'dtype=object' when creating the ndarray

Expected but not happened: Can not see the animation output in the python notebook

AtsushiSakai, 2021-09-29:
Thank you for reporting. Can you create a PR for it?

kajal-puri, 2021-10-18:
Hi, if this issue isn't solved. Can you assign this to me?

AtsushiSakai, 2021-11-12:
Close by #552

closed 2021-11-12 by AtsushiSakai