AtsushiSakai/PythonRobotics #547: CppRobotics, you inspired me with PythonRobotics!

giacomo-b, 2021-09-27:
First of all I want to say that I love this repo. I've been browsing it for nearly three years and I found it extremely useful for my thesis as well. I also read your paper mentioned in the readme. So thank you for all the hard work!

I recently decided to start working on [CppRobotics](https://github.com/giacomo-b/CppRobotics), inspired by this repo.

The goal is not to port the code as it is now (and I am not interested in just porting the algorithms from one language to another), but to create a "modular" library that allows any system (linear or nonlinear) to interact with any algorithm available(control, estimation, etc.)

Also, I want the library to be fast (it's fully templated to make use of static-size vectors and matrices) and easy to use (I am keeping it header-only and with only minimal dependencies). This would allow it not only to be educational, but also to be used in practical applications as it is.

I just started working on it, and it is currently a one-person effort, so it will take a while before it gets to "version 1.0". At the moment, there's only a PID, an LQR, and an EKF, which still need some additional work. However, I am determined to keep working on it in my free time and any contribution is more than welcome.

If you have any thoughts or suggestion to share, I would really appreciate it. Also, if you like the idea and/or you think the project could be useful, please leave a star. It would mean the world!

A final note: I mentioned this repo multiple times in README.md, but let me know if you would like this repo to be credited in any other way!

AtsushiSakai, 2021-09-29:
Thank you for your kind comments. I love your project, I added a star for it : ).

> I mentioned this repo multiple times in README.md, but let me know if you would like this repo to be credited in any other way!
I think it would be perfect if you add a reference to this repo in your LICENSE file.

giacomo-b, 2021-09-29:
> I love your project, I added a star for it : ).

Thank you, I appreciate this!

> I think it would be perfect if you add a reference to this repo in your LICENSE file.

Absolutely, thank you for mentioning it. I updated the [LICENSE](https://github.com/giacomo-b/CppRobotics/blob/master/LICENSE) file with the [standard way of crediting portions of the code](https://softwareengineering.stackexchange.com/a/277699), please check it out and see if it works for you. :)

AtsushiSakai, 2021-09-30:
It looks OK. Thanks!!.

closed 2021-09-30 by AtsushiSakai