AtsushiSakai/PythonRobotics #550: RRT* with reeds-sheep seems to produce illegal paths

davidenitti, 2021-10-02:
**Describe the bug**
Sometimes the path produced by RRT* with reeds-sheep script rrt_star_reeds_shepp.py seems not allowed: there are sharp corners in the path.

**Expected behavior**
allowed path

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - Python version 3.8.10

AtsushiSakai, 2021-10-03:
Thank you. Can you share your code to reproduce the issue?

davidenitti, 2021-10-03:
the code is rrt_star_reeds_shepp.py unmodified. However, this issue does not always appears, given the randomness of RRT, so I have to see if I manage to put a seed to the randomness to reproduce the problem. I'll let you know if I manage to do this.

davidenitti, 2021-10-03:
so I managed to reproduce the issue using the script rrt_star_reeds_shepp.py and setting seed 8 after random import:
```
import random
random.seed(8)
```
I get this:
[image]

a detail
[image]

davidenitti, 2021-10-08:
if you find a way to fix it let me know, thanks!

RyderCRD, 2023-04-01:
Please let me take this issue as part of my course project on SUSTech EEE5058. 

I found that the direct reason for this unexpected result is the difference between the length of the node's x_list and yaw_list. So when you zip() lists with different lengths, the longer parts of the longer lists was discarded.

Noticed that in this random.seed(8) case, RRTStarReedsShepp.planning() gives 101 nodes (in self.node_list), and 21 of these nodes have different lengths on the node.x_list (or y_list) and the node.yaw_list. Some of these nodes' yaw_lists are longer than the corresponding x_lists, and in this case, it's fine since all the x and y can be zipped and plotted correctly. However, when the yaw_lists are shorter than the x_lists, some x and y are discarded by zip() so some points on the curve are missed in the final plot.

A simple way to get a good figure is to add elements to the yaw_list to make it as long as the x and y lists.

[image]

[image]

It's strange that when adding new_nodes to the node_list, all nodes seem to have the same length on these lists.
But when we check the node_list after planning, the lengths became different.

Now I'm working on what makes the lengths of these lists different, and I will fix it soon.

AtsushiSakai, 2023-04-01:
Thank you. Your PR is welcome.

PR #812 (2023-04-27): Fix the rewire() of rrt* path planning (#550) 

#### Reference issue

fix #550 

#### What does this implement/fix?

rrt_star_reeds_shepp.py needs its own version of rewire(), which not only updates path_x , path_y, but also path_yaw when rewiring. 
Before this pull request, it uses the rewire() in rrt_star.py, which only cares about path_x, and path_y, thus issue #550 might occur.

The range of phi in rrt_star_reeds_shepp.py is changed from (0, 0.99\*math.pi) to (0.01\*math.pi, 0.99*math.pi) for the sake of speed.

#### Additional information

Two tests are added into tests/test_rrt_star_reed [... cut]
commit 80124f20 (2023-04-27): Fix the rewire() of rrt* path planning (#550)  (#812)

* Update rrt_star_reeds_shepp.py to fix #550

Add step_size attribute to RRTStarReedsShepp, and a method set_random_seed() to set the random seed, with two test cases.

* Update rewire() of rrt_star.py

Update rewire() of rrt_star.py to fix #550.  Since the old version didn't assign path_yaw of edge_node to near_node, the old rewire() doesn't fit rrt_star_reeds_shepp.py

* Update reeds_shepp_path_planning.py

Limit the range of phi for the sake of planning speed, and simplify the the calculation process in straight_left_straight().

* Upda [... cut]
closed 2023-04-27 by AtsushiSakai