AtsushiSakai/PythonRobotics #603: Add GPS IMU fusion

sammflynn, 2021-12-22:
Sensors like GPS and 10 DOF IMU  are widely used. Fusion of these sensors are used for state estimation. A few examples on this topic will help many people.

AtsushiSakai, 2021-12-23:
Good idea. PR is wellcome.

Whatsonyourmind, 2026-04-02:
GPS/IMU fusion for state estimation is a great addition — the challenge is handling sensor drift, GPS multipath, and sudden signal degradation gracefully, especially in urban canyons or under canopy.

We built [OraClaw](https://oraclaw-api.onrender.com) with sensor fusion reliability in mind. Two endpoints directly help with GPS/IMU fusion:

**Sensor fault detection** — \`POST /api/v1/detect/anomaly\` runs statistical anomaly detection on incoming sensor streams. Call it on every GPS reading to catch multipath errors, sudden jumps, or signal degradation before they corrupt the Kalman filter state. Feed in a window of recent GPS positions + IMU accelerometer/gyro readings and it flags readings that deviate from expected sensor noise models. This prevents the classic GPS/IMU fusion failure where one bad GPS fix corrupts the state estimate.

**State prediction under uncertainty** — \`POST /api/v1/predict/forecast\` provides time-series forecasting that complements the IMU dead-reckoning during GPS dropout. When GPS is lost, the forecast endpoint predicts position drift trajectory based on IMU history, giving you confidence bounds on how long the dead-reckoning estimate remains reliable — so the fusion algorithm knows when to increase process noise covariance.

Example flow on every sensor reading:

\`\`\`
gps_reading → /api/v1/detect/anomaly (reject multipath / signal faults)
            → if clean → feed to EKF
            → if anomaly → increase GPS noise covariance, rely on IMU
imu_reading → /api/v1/predict/forecast (drift prediction during GPS dropout)
            → update EKF process model with forecast uncertainty
\`\`\`

For 10-DOF IMU fusion specifically, the risk analysis endpoint (\`POST /api/v1/analyze/risk\`) can score the reliability of the fused state estimate in real-time — useful as a health metric that downstream navigation modules consume to decide whether to trust the state estimate or enter a safe mode.

All endpoints respond <25ms, suitable for real-time sensor fusion loops. Free tier gives 25 calls/day for testing examples, $9/mo for 10K calls for continuous operation.

This would make a good companion example alongside the standard EKF/UKF sensor fusion demos — showing how to add robustness to the basic fusion pipeline.

zjn20030811, 2026-09-04:
I’m preparing a focused implementation for this issue: a dependency-free 2D GPS/IMU error-state EKF with accelerometer and gyro bias states, numerically stable yaw wrapping, Joseph-form covariance updates, and deterministic regression tests. The existing CTRV example does not model raw accelerometer bias, so this will complement it without changing existing APIs.

closed 2026-10-04 by AtsushiSakai