AtsushiSakai/PythonRobotics #644: HybridAStar: Calculation of Buggle radius

trickster-117, 2022-02-15:
**Describe the bug**
The radius of the bubble containing a rectangular vehicle is:
 r = sqrt( (L/2)² + (W/2)²)
In the code it is:
r = sqrt( (L/2)² + 1)
See: https://github.com/AtsushiSakai/PythonRobotics/blob/e494ed6eaecb00fdf13fb361db947498c70d898c/PathPlanning/HybridAStar/car.py#L16 
This is valid given the default width of W=2. but not if the width is changed to another value.

**Expected behavior**
The formula should be changed as shown.

AtsushiSakai, 2022-02-24:
@trickster-117 Thank you for reporting. PR is welcome.

vidit21srivastava, 2022-03-23:
Is this issue still open for working upon?
I would like to work on this issue.

trickster-117, 2022-03-23:
Yes, this issue is still open, I haven't fixed it yet, but I will do it today. Further I found some improvements that would be nice to implement. I will try to implement them today.

closed 2022-04-11 by AtsushiSakai