AtsushiSakai/PythonRobotics #696: DOC: enhance Circle fitting doc

AtsushiSakai, 2022-07-19:
https://atsushisakai.github.io/PythonRobotics/modules/mapping/rectangle_fitting/rectangle_fitting.html

Abelarm, 2022-10-01:
Hi @AtsushiSakai,

I’d like to take this! could you please point to what you would like to be added ?
An example of a “good” doc would be great.

Thanks

AtsushiSakai, 2022-10-09:
Hi. Thank you for your proposal. [This code](https://github.com/AtsushiSakai/PythonRobotics/blob/master/Mapping/rectangle_fitting/rectangle_fitting.py) is based on this paper: [Efficient L\-Shape Fitting for Vehicle Detection Using Laser Scanners \- The Robotics Institute Carnegie Mellon University](https://www.ri.cmu.edu/publications/efficient-l-shape-fitting-for-vehicle-detection-using-laser-scanners/)
We needed to add descriptions of how the algorithm detects the rectangle based on the paper (eg. mathematics description about the AREA, CLOSENESS, VARIANCE criteria, etc.)
I think [this doc of histogram filter](https://atsushisakai.github.io/PythonRobotics/modules/localization/histogram_filter_localization/histogram_filter_localization.html) is an good example.

PR #848 (2023-06-30): Enhance rectangle fitting docs

#### Reference issue

fix #696

#### What does this implement/fix?

#### Additional information

#### CheckList
- [ ] Did you add an unittest for your new example or defect fix?
- [ ] Did you add documents for your new example?
- [ ] All CIs are green? (You can check it after submitting)
closed 2023-06-30 by AtsushiSakai