AtsushiSakai/PythonRobotics #712: 'float' object has no attribute 'rot_mat_2d'

manishsaini0901, 2022-08-24:
**Describe the bug**
The extended Kalman filter python file has a bug in line 156. Where in "fx = angle.rot_mat_2d(angle) @ (np.array([x, y]))", "angle" needs to be removed as we are directly importing the "rot_mat_2d" from utils.angle module. So, after removing it will become: "fx = rot_mat_2d(angle) @ (np.array([x, y]))"

**Expected behavior**
After correcting, it should not produce an error.

**Desktop:**
 - Python version 3.10.
 - OS version ubuntu 20.04

manishsaini0901, 2022-08-24:
It's the first ever issue that I have generated. Please correct me wherever I am wrong.

AtsushiSakai, 2022-08-24:
Can you check the latest code? Already, there is no `angle`.
https://github.com/AtsushiSakai/PythonRobotics/blob/ba0a5106d9c35600504bc338cfc4746247485956/Localization/extended_kalman_filter/extended_kalman_filter.py#L156

manishsaini0901, 2022-08-28:
Is it a temporary branch? I cloned the code from the master branch and it had this error.

AtsushiSakai, 2022-09-01:
No. Please check the master branch coce:https://github.com/AtsushiSakai/PythonRobotics/blob/master/Localization/extended_kalman_filter/extended_kalman_filter.py

closed 2022-09-07 by AtsushiSakai