AtsushiSakai/PythonRobotics #761: rot_mat_2d: scipy vs numpy

Alexbeast-CN, 2022-12-15:
I write a short script to test the speed of two versions of rot_mat_2d() function. The one used in PythonRobotics is rot_mat_2d(). It turns out numpy version is faster than scipy. So, I wander why using scipy to write this function? It's not as explicit, and fast as numpy.
```py
from scipy.spatial.transform import Rotation as Rot
import numpy as np

def rot_mat_2d(angle):
    return Rot.from_euler('z', angle).as_matrix()[0:2, 0:2]

def rot_mat_2d2(angle):
    return np.array([[np.cos(pi), -np.sin(pi)],
                     [np.sin(pi), np.cos(pi)]])

import time
t = 100000
cur_time = time.time()
for _ in range(t):
    rot_mat_2d(pi)
print(cur_time - time.time())

cur_time = time.time()
for _ in range(t):
    rot_mat_2d2(pi)
print(cur_time - time.time())
```

The output is:
```
-0.6680886745452881
-0.3319261074066162
```

AtsushiSakai, 2022-12-31:
Actually, no big reason. I’m a maintainer of SciPy, So I prefer to use SciPy one, if possible.

Alexbeast-CN, 2023-01-03:
Got it. Cheers!

closed 2023-01-03 by Alexbeast-CN