AtsushiSakai/PythonRobotics #877: move_to_pose.py not consistent with book when alpha > pi/2

qlibp, 2023-07-17:
**Describe the bug**
As the book says,
> For the case where the goal is behind the robot, that is α ∉ (−pi/2 , pi/2 ], we reverse the vehicle by negating v and γ in the control law

But the implementation here just negate linear velocity.

**Expected behavior**
I tested with negating both linear velocity & angular velocity, the control vibrates, but the overall trajectory makes more sense than only negating the linear velocity.

**Screenshots**
1. Negate Both
[image]

2. Negate only linear velocity
[image]

gongbingyu, 2024-12-05:
@qlibp I agree with this view. I have also tried this measure, but there is a vibration situation. Can you tell me how you suppress vibration? Can you share that part of the code?

Aglargil, 2025-02-19:
> [@qlibp](https://github.com/qlibp) I agree with this view. I have also tried this measure, but there is a vibration situation. Can you tell me how you suppress vibration? Can you share that part of the code?

Simply changing the sign of w to -w when alpha > pi/2 or alpha < -pi/2 is not enough, because alpha may jump between -pi and pi at this time, causing the sign of w to also jump, resulting in vibration. Therefore, when alpha > pi/2 or alpha < -pi/2, alpha needs to be recalculated using `alpha = angle_mod(np.arctan2(-y_diff, -x_diff) - theta)`.

PR #1168 (2025-02-20): Update move_to_pose for cases where alpha > pi/2 or alpha < -pi/2

#### Reference issue

fix #877

#### What does this implement/fix?

- Update move_to_pose for cases where alpha > pi/2 or alpha < -pi/2
- Format move_to_pose.py using Ruff

#### Additional information

Current results:
[image]

Previous results(master):
[image]

#### CheckList
- [x] Did you add an unittest for your new example or defect fix?
- [ ] Did you add documents for your new example?
- [x] All CIs are green? (You can check it after submitting)
closed 2025-02-20 by AtsushiSakai