AtsushiSakai/PythonRobotics #878: move_to_pose.py result not consistent with book

qlibp, 2023-07-17:
**Describe the bug**
move_to_pose.py simulation result not consistent with book.
The trajectory is obviously not consistent with the book, fig. 4.14

**Expected behavior**
The trajectory should at least look similar to book's simulation result.

**Screenshots**
1. Books' simulation result
[image]
2. code simulation result(with same param kpho=3, kalpha=8, kbeta=3)
Note that I have already consider the case that `kbeta_book = -kbeta_code`

[image]

us107, 2024-10-15:
Hi! I'd love to contribute to this issue. Could you please assign it to me?

AtsushiSakai, 2024-10-15:
PR is welcome, no need to assignment

us107, 2024-10-15:
Thank you so much!