AtsushiSakai/PythonRobotics #907: grid_based_sweep_coverage_path_planner.py last example the region coverage problem

kk-1, 2023-09-18:
**Describe the bug**
The last region example in the code:
PathPlanning/GridBasedSweepCPP/grid_based_sweep_coverage_path_planner.py

I mean the last example:
    ox = [0.0, 20.0, 50.0, 200.0, 130.0, 40.0, 0.0]
    oy = [0.0, -80.0, 0.0, 30.0, 60.0, 80.0, 0.0]
    resolution = 5.0
    planning_animation(ox, oy, resolution)

does not produce complete coverage on my system settings.

**Expected behavior**
A clear and concise description of what you expected to happen.

The coverage is not like in the animated gif shown:
https://github.com/AtsushiSakai/PythonRoboticsGifs/raw/master/PathPlanning/GridBasedSweepCPP/animation.gif

**Screenshots**
If applicable, add screenshots to help explain your problem.
Check the following image:

[image]

**Desktop (please complete the following information):**
I used Python 3.10.13 on Linux (opensuse tumbleweed)
numpy                             1.25.2
matplotlib                        3.6.3
matplotlib-inline                 0.1.6
scipy                             1.11.1

Can you let me know which python and lib versions works for that example.

Thanks.

UjjwalPardeshi, 2023-10-01:
can you assign me this please

kk-1, 2023-10-02:
Further investigation suggests float2integer rounding issue.
Probably different Python/package versions have different precision defaults.

AtsushiSakai, 2023-10-15:
The latest codes have still this issue? I merged this PR https://github.com/AtsushiSakai/PythonRobotics/pull/917

kk-1, 2023-10-18:
Thanks it works now!

closed 2023-10-18 by AtsushiSakai