AtsushiSakai/PythonRobotics #915: Fixing grid_based_sweep_coverage_path_planner.py

ssr-yuki, 2023-10-04:
Hi.

The current behavior of `PathPlanning/GridBasedSweepCPP/grid_based_sweep_coverage_path_planner.py` is different from one shown in the [doc](https://atsushisakai.github.io/PythonRobotics/modules/path_planning/coverage_path/coverage_path.html).

**Screenshots**

| current | expected (same as shown in the [doc](https://atsushisakai.github.io/PythonRobotics/modules/path_planning/coverage_path/coverage_path.html)) |
| :---: | :---: |
| [image] | [image] |
| [image] | [image] | 

**Cause**

Investigating [the change log](https://github.com/AtsushiSakai/PythonRobotics/commits/master/PathPlanning/GridBasedSweepCPP/grid_based_sweep_coverage_path_planner.py), I found that [the latest commit](https://github.com/AtsushiSakai/PythonRobotics/commit/0fc769421f66a6403f750781dc46534d84e7c424) caused this behavior change. 

[image]

The change in L54-55->L53 of the file (also see the image) silently transformed the `occupied_val` argument of `check_occupied_from_xy_index` function from 1.0 into 0.5.

As a reference, the following modification for the current L53 can correct this issue temporally.

```python
-                 if self.check_occupied(next_c_x_index, next_c_y_index, grid_map):
---
+                 if grid_map.check_occupied_from_xy_index(next_c_x_index,
                                                           next_c_y_index, FloatGrid(1.0)):
```

Please check it. Thanks.

MrDavidAlv, 2023-10-12:
good

AtsushiSakai, 2023-10-15:
Thank you for great explanation about the issue.

PR #917 (2023-10-15): Revert a parameter in grid_based_sweep_coverage_path_planner to correct the planning results

#### Reference issue

This PR resolves #915 .

#### What does this implement/fix?

This PR just reverts the undesirable parameter change in `PathPlanning/GridBasedSweepCPP/grid_based_sweep_coverage_path_planner.py` summarised in #915 .

#### Additional information

N/A

#### CheckList

- [x] Did you add an unittest for your new example or defect fix?
- [ ] Did you add documents for your new example?
- [x] All CIs are green? (You can check it after submitting)
closed 2023-10-15 by AtsushiSakai