AtsushiSakai/PythonRobotics #944: Strange Animation Results in test_rrt_star_reeds_shepp.py

videh25, 2023-12-01:
### Bug
test3 in test_rrt_star_reeds_shepp.py tries to generate a path with step_size = 20. Since it is a test file, animation is kept off. However, on turning the animation on, the result is very  weird. RRT does not plan with smooth RS curves but rather with straight lines with sharp corners.

### Expected Behaviour
Also ran test2 from test_rrt_star_reeds_shepp.py (step_size = 0.2). They plan and generate smooth RS curves only. 
 
### Screenshots
**test3 Animation**
[image]

***test2 Animation***
[image]

### Desktop
 - Python 3.9.5
 - numpy 1.24.3
 - scipy 1.10.1
 - matplotlib 3.7.1
 - cvxpy 1.4.1
 - Ubuntu 20.04

AtsushiSakai, 2024-01-08:
@videh25 Is it OK to close this issue by your PR?

videh25, 2024-01-08:
Yes, sounds great!

closed 2024-01-08 by videh25