AtsushiSakai/PythonRobotics #966: ModuleNotFoundError: No module named 'utils'

rosrookie, 2024-01-10:
Hi, when I git clone the whole project. and installed all requirements. And when I run with mpc. error happends:

`ModuleNotFoundError: No module named 'utils'`

I do not change anything, just clone and run, then error happends.

rosrookie, 2024-01-10:
well, it seems like I change the code order, then wroked.

before:
`
import matplotlib.pyplot as plt
import cvxpy
import math
import numpy as np
import sys
import pathlib
from utils.angle import angle_mod
sys.path.append(str(pathlib.Path(__file__).parent.parent.parent))

from PathPlanning.CubicSpline import cubic_spline_planner
`

after:
`
import matplotlib.pyplot as plt
import cvxpy
import math
import numpy as np
import sys
import pathlib
sys.path.append(str(pathlib.Path(__file__).parent.parent.parent))

from PathPlanning.CubicSpline import cubic_spline_planner
from utils.angle import angle_mod
`

rosrookie, 2024-01-10:
it seems like when I change the order, then worked.

before:

```
import matplotlib.pyplot as plt
import cvxpy
import math
import numpy as np
import sys
import pathlib
from utils.angle import angle_mod
sys.path.append(str(pathlib.Path(__file__).parent.parent.parent))

from PathPlanning.CubicSpline import cubic_spline_planner
```

after:
```
import matplotlib.pyplot as plt
import cvxpy
import math
import numpy as np
import sys
import pathlib
sys.path.append(str(pathlib.Path(__file__).parent.parent.parent))

from PathPlanning.CubicSpline import cubic_spline_planner
from utils.angle import angle_mod
```

closed 2024-01-10 by rosrookie

ZhaiMen-Hub, 2024-04-01:
Find the same problem in lqr_speed_steer_control.py

kamocat, 2024-11-03:
This seems to be a recurring issue:

- #709 
- #750 
- #754 

Either we should fix all the scripts as above, or update the "HOW TO USE" section with instructions to install the 'utils' module.