AtsushiSakai/PythonRobotics #995: Plot issue

Ombenavides, 2024-03-25:
**Describe the bug**
I am quite a noob at this kind of request so please forgive me if I make mistakes :)
I've been trying to run the file random_inverse_kinematics after installing everything and also importing the lib. But my code just plots blank. I cannot see anything. :( 

**Expected behavior**
I expected to get the plot for the random kinematics to explore the code

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - Python version 3.12.2
 - Each library version 
 - OS version Windows 11 home

Maybe I am committing a basic mistake. But please help me out with this issue. 
Thanks in advance.

GiladRobee, 2024-04-10:
Not an expert either so take it with a grain of salt.
I found out NLinkArm is using mpl_toolkits which depend on old Matplotlib functions that were deprecated some versions ago.
try and use it in a Venv, docker with the right lib version (probably the ones from the requirement folder.

AtsushiSakai, 2024-04-28:
@Ombenavides Hi. Thank you for reporting. I can reproduced the issue. PR #1007 should fix this issue. Please try it with the latest code.

PR #1007 (2024-04-28): fix plot issue in random_inverse_kinematics.py

#### Reference issue

Fix #995 

#### What does this implement/fix?

#### Additional information

#### CheckList
- [ ] Did you add an unittest for your new example or defect fix?
- [ ] Did you add documents for your new example?
- [ ] All CIs are green? (You can check it after submitting)
closed 2024-04-28 by AtsushiSakai