AtsushiSakai/PythonRobotics #1073: Extend frenet_optimal_trajectory to support more scenarios

Aglargil, 2024-09-11:
Hi there,
First of all, thank you for the amazing PythonRobotics library! It's an incredibly valuable resource for learning robotics algorithms.
I've been working on the frenet_optimal_trajectory implementation lately, and noticed that it currently only handles **Velocity Keeping** for **High Speed Trajectories**.
I'm interested in extending it to support additional scenarios mentioned in the paper, such as:
- **Low Speed Trajectories**
- **Following, Merging, and Stopping**

Would you be open to accepting contributions to frenet_optimal_trajectory to handle these cases? I'd be happy to submit PR to implement them.

AtsushiSakai, 2024-12-21:
@Aglargil Thank you. Yes, any kind of enhancement PRs are welcome.

Aglargil, 2024-12-24:
@AtsushiSakai 
Hello, I have submitted a PR addressing the content mentioned above: https://github.com/AtsushiSakai/PythonRobotics/pulls.

If there are any areas that need improvement, please feel free to let me know, and I will address them promptly.

Lastly, I am truly delighted to contribute in a small way to this amazing project. Wishing you a wonderful day!

PR #1114 (2025-01-22): Extend frenet_optimal_trajectory to support more scenarios
#### Reference issue
fix #1073

#### What does this implement/fix?

- Add `CartesianFrenetConverter`: A class for converting states between Cartesian and Frenet coordinate systems
- Add third derivative and curvature rate calculations to `CubicSpline`
- Add LowSpeed/HighSpeed `LateralMovementStrategy`, add MergingAndStopping/VelocityKeeping `LongitudinalMovementStrategy`
- Fix : `s_d` was used as the acceleration, but since` s_d` only represents the longitudinal acceleration, the `CartesianFrenetConverter` is now used to compute the ac [... cut]
closed 2025-01-22 by AtsushiSakai