AtsushiSakai/PythonRobotics #1219: Add VSLAM or 3D lidar SLAM algorithms

devyani-code, 2025-05-14:
Visual SLAM and 3D Lidar SLAM algorithms are widely used in the industry. Concepts like visual odometry, object tracking using 3D lidars, etc., can be helpful to many people.

AtsushiSakai, 2025-05-25:
I think it’s a good idea. However, it might be difficult to implement sample code for 3D LIDAR or Visual SLAM without using any external libraries... But PR is welcome.

kartikmadhira1, 2025-05-26:
I think adding core concepts in 3D computer vision first would be a good starting point - Stereo, E and F matrices, PnP, Triangulation and Bundle Adjustment without libraries. These can then be re-arranged for a bigger VSLAM module.

devyani-code, 2025-08-25:
Okay! Will start with Camera geometery basics and stereo vision.

rin-katsuragi, 2026-10-10:
Hey @AtsushiSakai is this still open? Can I get involved?